hdl-registers/hdl-registers #116: Make it possible to group fields of different types

LukasVik, 2024-05-10:
Or, in other words, fully preserve the order of register and fields, so they appear in the `RegisterList` object exactly the way the appeared in the data file (TOML/JSON/YAML).

# Motivation

Grouping things that are related is very important for readability and understandability. Both in data file, HDL, SW and documentation. 

Our current data file format and parser will preserve the order of fields, but only within each type. Bits will always be placed first, then all bit vectors, then all integers, and then all enumerations. No matter how they appeared in the data file. Meaning, it is impossible to group, e.g., a bit with a bit vector.

Same on the top level, all registers are placed first then all register arrays. No matter how they appeared in the data file.

## Example

Say that a register has the following fields:

- **tuser_enable** (bit)
- **tuser_value[3:0]** (bit vector)
- **tdest_enable**
- **tdest_value[3:0]**
- **tid_enable**
- **tid_value[3:0]**

It would be really nice if all the **TUSER** fields were close together, and same for the **TDEST** and **TID** fields. This is currently not possible.

# Current problem description

The current data file format organizes data e.g. like this:

**register**._register_name_.mode

**register**._register_name_.**bit**._field_name_1_.description
**register**._register_name_.**bit**._field_name_1_.default_value

**register**._register_name_.**integer**._field_name_2_.description
**register**._register_name_.**integer**._field_name_2_.default_value

**register**._register_name_.**bit**._field_name_3_.description
**register**._register_name_.**bit**._field_name_3_.default_value

The result from reading a TOML/JSON/YAML file in Python is a dictionary. To append a field to a register, we must parse name, description, etc. from the dictionary. In order to access these properties, we must first enter and iterate the third level of the dictionary (**bit**/**integer**/etc). This is were the ordering of fields is lost, since we must decide which one to enter first, and then parse all fields in that type. There is no way to know that the first bit shall be placed first, then the first integer field, and then the second bit.

# Proposed solution

Given the current problem description above, it seems the data file format must be updated. This is a highly breaking change, and will be a hassle for many users. However, we will be able to convert old data files to the new format automatically. The hassle will be that any manual formatting of the old file will be lost. As well as any comments.

I will suggest two new formats here. First in pseudo-code as above, and then after that with some real TOML code, using some different TOML formatting.

## New format A: Type from structure

Simply flip the order compared to the current format. Allows us to iterate all members of the register in order, no matter their type.

_register_name_.**register**.mode

_register_name_.**register**._field_name_1_.**bit**.description
_register_name_.**register**._field_name_1_.**bit**.default_value

_register_name_.**register**._field_name_2_.**integer**.description
_register_name_.**register**._field_name_2_.**integer**.default_value

_register_name_.**register**._field_name_3_.**bit**.description
_register_name_.**register**._field_name_3_.**bit**.default_value

## New format B: Type from property

Simply flip the order compared to the current format. Allows us to iterate all members of the register in order, no matter their type.

_register_name_.type = "**register**"
_register_name_.mode

_register_name_._field_name_1_.type = "**bit**"
_register_name_._field_name_1_.description
_register_name_._field_name_1_.default_value

_register_name_._field_name_2_.type = "**integer**"
_register_name_._field_name_2_.description
_register_name_._field_name_2_.default_value

_register_name_._field_name_3_.type = "**bit**"
_register_name_._field_name_3_.description
_register_name_._field_name_3_.default_value

# Sample data files

## New format A

```TOML
################################################################################
[data.register]

mode = "w"

################################################################################
[status.register]

mode = "r_w"
description = "Status register"

bad.bit.description = "Bad things happen"

not_good.bit.description = ""
not_good.bit.default_value = "1"

direction.enumeration.description = "The direction mode"
direction.enumeration.default_value = "input"
direction.enumeration.element.passthrough = ""
direction.enumeration.element.input = ""
direction.enumeration.element.output = "use in output mode"

interrupts.bit_vector.width = 4
interrupts.bit_vector.description = "Many interrupts"
interrupts.bit_vector.default_value = "0110"

count.integer.description = "The number of things"
count.integer.min_value = -5
count.integer.max_value = 15

################################################################################
[config.register_array]

array_length = 3
description = "A register array"

# ------------------------------------------------------------------------------
[config.register_array.input_settings.register]

description = "Input configuration"
mode = "r_w"

[config.register_array.input_settings.register.enable.bit]

description = "Enable things"
default_value = "1"

[config.register_array.input_settings.register.disable.bit]

description = ""
default_value = "0"

[config.register_array.input_settings.register.number.integer]

description = "Configure number"
max_value = 3
default_value = 1

[config.register_array.input_settings.register.size.enumeration]

element.small = ""
element.large = ""

# ------------------------------------------------------------------------------
[config.register_array.output_settings.register]

mode = "w"

[config.register_array.output_settings.register.data.bit_vector]

width = 16
description = "Some data"
default_value = "0000000000000011"
```

## New format B

```TOML
################################################################################
[data]

type = "register"
mode = "w"

################################################################################
[status]

type = "register"
mode = "r_w"
description = "Status register"

bad.type = "bit"
bad.description = "Bad things happen"

not_good.type = "bit"
not_good.description = ""
not_good.default_value = "1"

direction.type = "enumeration"
direction.description = "The direction mode"
direction.default_value = "input"
direction.element.passthrough = ""
direction.element.input = ""
direction.element.output = "use in output mode"

interrupts.type = "bit vector"
interrupts.width = 4
interrupts.description = "Many interrupts"
interrupts.default_value = "0110"

count.type = "integer"
count.description = "The number of things"
count.min_value = -5
count.max_value = 15

################################################################################
[config]

type = "register array"
array_length = 3
description = "A register array"

# ------------------------------------------------------------------------------
[config.input_settings]

type = "register"
description = "Input configuration"
mode = "r_w"

[config.input_settings.enable]

type = "bit"
description = "Enable things"
default_value = "1"

[config.input_settings.disable]

type = "bit"
description = ""
default_value = "0"

[config.input_settings.number]

type = "integer"
description = "Configure number"
max_value = 3
default_value = 1

[config.input_settings.size]

type = "enumeration"
default_value = "large"
element.small = ""
element.medium = ""
element.large = ""

# ------------------------------------------------------------------------------
[config.output_settings]

type = "register"
mode = "w"
```

# Questions for the community

This is a large change that will impact almost all users. So I would appreciate some feedback and help on this one. This is hopefully the last change of data file format we ever make. 

1. Do you prefer format A or format B above, or anything else?
2. If B, do you prefer type names written with space (**bit vector**, **register array**) or underscore (**bit_vector**, **register_array**)?
3. Any other wishes for the data file format? Anything that is missing? Anything that is strange/wrong?

LukasVik, 2024-05-10:
I have implemented and tested option A and B in #115. They both work fine.

LukasVik, 2024-05-10:
Pining some people that might be interested: @LarsAsplund @elhasab @codecache-fpga @joafr-ice.

Is also discussed in #95.

LukasVik, 2024-05-10:
@theo-scott @woloven70 @hansfbaier @pidgeon777 @rkaletka @OldTomCrazyCode  @basberg @eddiell @wlupo @sduport @pcayuela  @mustafatayf @mariuselv @prasadp4009 @robindust-ce @tms4517 @akashlevy @svenka3 @pintert3 @lgrativol @nicdes

Sorry for the mass ping of everyone who has either starred or forked this project. But if you are interested in the future of this project, this is the time to make your voice heard!

pintert3, 2024-05-10:
I haven't yet used this tool much, but from what I see, format B seems the better option. It's clearer and easier to write. For the names, I prefer the underscore.

pidgeon777, 2024-05-10:
To be honest I hadn't a chance to test this tool yet, but I plan to do it as soon as possible 👍.

rkaletka, 2024-05-10:
for me format B with underscore, so far no wishes, gotta work more with it

elhasab, 2024-05-12:
I've been pondering this issue and wondered if it's feasible to maintain the file format unchanged while enforcing field creation by their order in the TOML file. However, I don't mind any of the proposed solutions.

As for my preference, I lean towards format B with underscore.

LarsAsplund, 2024-05-14:
B because it makes a clear distinction between the named objects that we talk about and properties of those objects.

Ideally, properties like types should be identifiers rather than strings but it when that doesn't work strings with underscores makes it more identifierish.

LukasVik, 2024-05-14:
Great, thank you all, it seems that option B with underscore is winning.

I think I have some time in the beginning of next week to implement this. Although it is mostly implemented #115, a lot of polishing, error checking, and unit testing remains. Once done I will make a release.

LukasVik, 2024-07-11:
This is now done and merged to main. I have some other changes I want to make before making a breaking release, so a proper release is probably a while away. But if anyone wants to try it from main in the meantime, feel free.

LarsAsplund, 2024-08-21:
@LukasVik I tested the updated format in my client's project and it works as intended! When do you think the next release will be? My client prefers waiting until it's available on PyPi. If the final release is far in the future, would you consider a [pre-release](https://packaging.python.org/en/latest/discussions/versioning) which only gets installed when doing `pip install --pre`?

I use that for VUnit. For example, the latest pre-release on PyPi is 5.0.0.dev3. The only "problem" is that this format doesn't follow semver which we normally use (some open source project ranking tools give extra points for following semver). For that reason the git tag is still v5.0.0-dev.3

LukasVik, 2024-08-21:
Hello @LarsAsplund , thank you for trying it out, nice to hear that it’s
working well.

I am ready to make a new release asap actually. I’ve done all the breaking
changes that I wanted to include with 6.0.0. I’m on the road now but I’ll
be back in Vienna on the 26th. I’ll make the release within a day or two
after that.

LukasVik, 2024-08-25:
This is now released in version 6.0.0!

LukasVik, 2024-08-25:
@LarsAsplund Thanks again for beta testing. Did the automatic upgrade of the data file work fine for you?