hdl-registers/hdl-registers #119: Minimal example incorrect?

cweickhmann, 2024-06-06:
**Description**
It seems, the minimal example under User Guide / Getting Started / Usage in the docs imports `from_toml` from the wrong place.

It says `from hdl_registers.parser import from_toml` but it should say `from hdl_registers.parser.toml import from_toml`.

Furthermore, it would be handy for a beginner to have the referenced `caesar_registers.toml` available. I cannot find such a file in the repository. Using the file e.g. [regs_test.toml](https://github.com/hdl-registers/hdl-registers/tree/main/tests/regs_test.toml) instead, or linking to it, would reduce confusion.

**How to reproduce?**

* Run `python3 -m pip install hdl-registers`
* Have `hdl_registers` version 5.2
* Paste example code and run it.

Observe complaint: `cannot import name 'from_toml' from 'hdl_registers.parser' (xxx/.local/lib/python3.10/site-packages/hdl_registers/parser/__init__.py)

* Fix by inserting `.toml` after `from hdl_registers.parser`.

**Context**
- Operating system: Ubuntu 22.04.4
- Python version: 3.10
- hdl-registers origin:
  - [x] Released package.
  - [ ] Git repository.
- hdl-registers version: 5.2.0

LukasVik, 2024-06-07:
@cweickhmann Thanks for pointing this out, fixed now!

closed 2024-06-07 by LukasVik

cweickhmann, 2024-06-10:
In the name of future-noob: Thank you! ;-)

commit d912c473 (2024-06-07): Fix minimal usage example

Closes #119
PR #120 (2024-06-07): Fix minimal usage example
Closes #119
commit b0631c13 (2024-06-07): Fix minimal usage example

Closes #119