hdl-registers/hdl-registers #213: TypeError: get_git_commit() got an unexpected keyword argument 'use_rst_annotation'

Botnic, 2025-03-11:
Hello Lukas

I am trying to evaluate hdl-registers. Your tool looks very promising. 

**Description**
I installed hdl-register over PIP, created a python script with the content from the documentation (see below) and copied the TOML file from the documentation next to it. 

When running the python script I the get the following message:
```
python3 gen_regs.py 
Creating VHDL register package file: test_reg_regs_pkg.vhd
Traceback (most recent call last):
  File "/data/home/reto.meier/Documents/olymp/scripts/hdl_registers/gen_regs.py", line 10, in <module>
    VhdlRegisterPackageGenerator(register_list=register_list, output_folder=this_dir).create_if_needed()
  File "/data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages/hdl_registers/generator/register_code_generator.py", line 204, in create_if_needed
    create_path = self.create(**kwargs)
                  ^^^^^^^^^^^^^^^^^^^^^
  File "/data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages/hdl_registers/generator/register_code_generator.py", line 167, in create
    result = f"{self.header}\n{code}"
                ^^^^^^^^^^^
  File "/data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages/hdl_registers/generator/register_code_generator.py", line 280, in header
    generated_source_info = self.comment_block(text=self.generated_source_info, indent=0)
                                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages/hdl_registers/generator/register_code_generator.py", line 290, in generated_source_info
    return self._get_generated_source_info(use_rst_annotation=False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages/hdl_registers/generator/register_code_generator.py", line 318, in _get_generated_source_info
    git_commit = get_git_commit(directory=directory, use_rst_annotation=use_rst_annotation)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: get_git_commit() got an unexpected keyword argument 'use_rst_annotation'
```

```Python
from pathlib import Path

from hdl_registers.generator.vhdl.register_package import VhdlRegisterPackageGenerator
from hdl_registers.parser.toml import from_toml

this_dir = Path(__file__).parent

register_list = from_toml(name="test_reg", toml_file=this_dir / "test_reg.toml")
VhdlRegisterPackageGenerator(register_list=register_list, output_folder=this_dir).create_if_needed()
```

**Context**
- Operating system: Redhat 8
- Python version: 3.11.0
- hdl-registers origin: Released package (pip)
- hdl-registers version: 7.1.0

**Additional context**
```
pip install hdl-registers -U
Requirement already satisfied: hdl-registers in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (7.1.0)
Requirement already satisfied: black in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from hdl-registers) (25.1.0)
Requirement already satisfied: docutils in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from hdl-registers) (0.21.2)
Requirement already satisfied: GitPython in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from hdl-registers) (3.1.44)
Requirement already satisfied: PyYAML in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from hdl-registers) (6.0.2)
Requirement already satisfied: rtoml in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from hdl-registers) (0.12.0)
Requirement already satisfied: termcolor in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from hdl-registers) (2.5.0)
Requirement already satisfied: tomli-w in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from hdl-registers) (1.2.0)
Requirement already satisfied: tsfpga>=12.3.6 in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from hdl-registers) (13.1.2)
Requirement already satisfied: vunit_hdl in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from tsfpga>=12.3.6->hdl-registers) (4.7.0)
Requirement already satisfied: click>=8.0.0 in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from black->hdl-registers) (8.1.7)
Requirement already satisfied: mypy-extensions>=0.4.3 in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from black->hdl-registers) (1.0.0)
Requirement already satisfied: packaging>=22.0 in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from black->hdl-registers) (24.0)
Requirement already satisfied: pathspec>=0.9.0 in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from black->hdl-registers) (0.12.1)
Requirement already satisfied: platformdirs>=2 in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from black->hdl-registers) (4.3.6)
Requirement already satisfied: gitdb<5,>=4.0.1 in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from GitPython->hdl-registers) (4.0.12)
Requirement already satisfied: smmap<6,>=3.0.1 in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from gitdb<5,>=4.0.1->GitPython->hdl-registers) (5.0.2)
Requirement already satisfied: colorama in /data/home/reto.meier/.pyenv/versions/3.11.0/lib/python3.11/site-packages (from vunit_hdl->tsfpga>=12.3.6->hdl-registers) (0.4.6)
```

LukasVik, 2025-03-11:
Hey @Botnic ,

That's a dependency pegging problem, sorry about that! I've made a new release where the tsfpga dependency is correctly pegged. You can update your installation with 

```
python3 -m pip install --upgrade hdl-registers
```

Running the example script again should work then.

LukasVik, 2025-03-11:
Feel free to re-open if the issue persists

Botnic, 2025-03-11:
Thanks works.