hdl-registers/hdl-registers #228: constants in default_registers

fpgauserdude, 2025-04-03:
**Is your feature request related to a problem? Please describe.**
I need to use default registers. One of those needs to be a constant. But it appears I cannot pass a constant in the default registers list.

**Describe the solution you'd like**
```
def default_registers(mm_file: Path) -> list:
    hsh = int(get_hash(mm_file)[0:8], 16)
    reg = Register(
        name='memmap_hash',
        index=0,
        mode=REGISTER_MODES['r'],
        description=f'read-only register of the MD5 hash of {mm_file.name}'
    )
    reg.append_integer(
        name='hash',
        description='',
        min_value=0,
        max_value=2**32-1,
        default_value=hsh
    )
    return [
        IntegerConstant(
            'MEMMAP_HASH',
            value=hsh,
            description=f'MD5 hash of {mm_file.name} ({hex(hsh)})'
        ),
        reg
    ]
```

**Describe alternatives you've considered**
I suspect I can use the python API to create the constant manually outside the TOML input file.  I will try that.

LukasVik, 2025-04-04:
The imagined use case for the `default_registers` argument to the parser is for a set of registers that shall be exactly the same in a number of register lists/modules. The feature comes from my first job where the embedded software's interrupt-based workflow was based on the first register of each module being `interrupt_status` and the second `interrupt_mask`. That's were I got the idea but I've honestly never used the mechanism.

For manipulating register/constant information on a single register list, the regular Python API can be used. I use that all the time. Sometimes what you want to do is easier to express in Python rather than TOML. Sometimes the information is only available in Python.

The return value of `from_toml` is a `RegisterList` (https://hdl-registers.com/rst/api_reference/hdl_registers.parser.html#hdl_registers.parser.toml.from_toml).
Meaning you can do `append_register` or `add_constant` on that object:

* https://hdl-registers.com/rst/api_reference/hdl_registers.html#hdl_registers.register_list.RegisterList.append_register
* https://hdl-registers.com/rst/api_reference/hdl_registers.html#hdl_registers.register_list.RegisterList.add_constant

Since I don't fully know you use case I can not say for sure, but it might be easier to manipulate the RegsiterList object like this, rather than to use the default registers mechanism.

LukasVik, 2025-04-09:
I personally add

1. Random build ID (integer)
2. Git sha at build time
3. Vivado version
4. Build hostname
5. Build timestamp
 
as register constants as part of the FPGA build flow. The build ID is also synthesized into a software-accessible register. The software can then check at runtime that the register build ID matches the constant in the header that the software was compiled with. If there is a mismatch, it should crash.

LukasVik, 2025-04-09:
The default register mechanism is needed because registers cannot be inserted in an existing and populated register list, they can only be appended at the end. This is by design. Constants on the other hand can be added at any point, with no functional difference. I think the inconsistency makes sense.

LukasVik, 2025-04-09:
In your case I think it would be more reliable to use a generic for the hash value and assign it in HDL to the read-only register. Rather than using the default value of the register. The default value will be part of the generated HDL, and will at some point be read by the synthesis process. But if you start another build in parallel, that might have a different default value. Or if you start a simulation.

LukasVik, 2025-05-19:
Closing since I think the questions has been answered. Feel free to reopen if there are further issues with this.