hdl-registers/hdl-registers #229: better SystemVerilog support

fpgauserdude, 2025-04-03:
**Is your feature request related to a problem? Please describe.**
We primarily use SystemVerilog, and I don't want to be constrained by the current limitations of the built-in SystemVerilog generator. I also don't want another package dependency.

**Describe the solution you'd like**
I propose to create a "struct_package.py" (analogous to "record_package.py" for VHDL). We would also need a new VHDL generator that would generate a wrapper around the VHDLAxiLiteWrapper such that the ports would be compatible with mixed language simulators (it appears there are already functions being generated in the VHDL package that convert the I/O records to std_logic_vectors). The AXI-lite records would have to broken out into discrete signals for mixed language instancing. Then generate a SystemVerilog wrapper around that VHDL wrapper and populate the generated structs generated structs appropriately, and probably also instances AXI-lite interface objects.

**Describe alternatives you've considered**
It might not be worth this effort, as there might be other register generator packages that natively support what I need. I am investigating alternatives.

**Additional context**
If we determine hdl-registers is the best option, I would be willing to add this functionality. The cost of this approach is that users would require a mixed language simulator (excludes open-source tools).

LukasVik, 2025-04-04:
I personally don't use SystemVerilog. I did the initial effort of adding support for the PeakRDL exporter, but I don't aim to do much more work than that myself. Instead community contributions from SystemVerilog users are welcome. I will of course be available for discussions, support, code reviews, etc. 

If the best solution is to ditch the PeakRDL exporter and write our own SV generator from the ground up, then I am in favor of that. But do note that some of the current limitations might actually be supported by PeakRDL, just that I didn't have the  time to implement them: https://hdl-registers.com/rst/generator/generator_systemverilog.html#details-and-limitations
The SystemRDL standard is a very intimidating 146 page document: https://www.accellera.org/images/downloads/standards/systemrdl/SystemRDL_2.0_Jan2018.pdf
Maybe there is support for these features in ther.

I am very hesitant towards a mixed-language solution that would lock in the user with a commercial simulator.

LukasVik, 2025-04-04:
The VHDL AXI-lite generator is a wrapper around this generic register file which uses SLV for register values: https://github.com/hdl-modules/hdl-modules/blob/main/modules/register_file/src/axi_lite_register_file.vhd
The wrapper simply converts the values from SLV to native field values in records. These same conversion functions are used in the simulation-support packages. 

https://hdl-registers.com/rst/generator/generator_vhdl.html

For my use case, this is a very powerful set of tools. The simulation packages save me so much time and effort everyday. I'm surprised there is no equivalent in SV generators.

LukasVik, 2025-04-09:
If there is work and effort put into making a custom SV generator I would prefer if it was accessible to everyone. In a perfect world. But I will not stop you if you think another solution is better for you. 

With that said, in my gut reaction I don't think there is much gain in wrapping the VHDL. Most of the complexity/value of the VHDL is in converting a list of SLV values into a high-level structure of nested records containing native values (enum/bool/float, etc).  I don't think any of that can be used in SV.