hdl-registers/hdl-registers #238: True Dual Port Ram

sigmuha, 2025-05-05:
Hi! 
I'm new to hdl-registers, and this looks like a very interesting project. :) 
One feature crucial feature for my projects are true dual port (tdp) rams. Are these supported in any way?

If not, maybe a keyword in the toml/yaml-file can indicate if this is to be put in such a memory block? 

I guess one challenge is that this is not vendor-agnostic, but this could possibly be solved through a cli-argument asking for a vendor type.

LukasVik, 2025-05-05:
Hey @sigmuha 

So you want the register values in fabric to be stored in BRAM instead of in FFs? So that instead of being able to access all values freely, you index a BRAM and in the next clock cycle you can use the register value?

Which HDL are you using? If VHDL, I think it should be quite possible.

sigmuha, 2025-05-05:
Yes! That is the correct use-case. 

I'm using VHDL.

LukasVik, 2025-05-09:
Also, the fact that you are emphasizing _true_ dual port. Do you want access the values in RAM in another clock domain than the register bus clock?

sigmuha, 2025-05-09:
Both modes would be great, but mostly «w», and «w_r» would be useful, as it is mostly used to set a configuration from sw to hw. But, reading back the values (mode «r») would also be very useful. :) 

Two separate clock domains would be nice, but not required.

LukasVik, 2025-05-19:
So the good news is that I don't think that any update to hdl-registers is needed. 

The "official" AXI-Lite register file (https://github.com/hdl-modules/hdl-modules/blob/main/modules/register_file/src/axi_lite_register_file.vhd) places register values in FFs. It is not really explicit, but that is most likely how it will synthesize. 

Implementing that writes are done to a BRAM instead would require an HDL update. You can copy and implement this update locally, and instantiate your own register file instead of the official. 

The ``reg_value`` signal and ``regs_down`` port would be replaced with a BRAM signal that is read and written synchronously. Possibly another BRAM, and replacement of `regs_up`, if "r"  mode register values should also be stored in BRAM. Setting the "RAM_STYLE" attribute to "BLOCK" should make it synthesize to block RAM. As long as both read and write are synchronous and happen to only one address at a time. Should make it quite portable, should not require any vendor IP or primitive.

I suspect that a register file implemented like this would not support the "r_w" mode. Since software readback of a previously written value would require a third BRAM port. Since read and write can happen in parallel.

I don't have much time to spare right now, and I don't have a personal interest in this feature, so I will not be able to help much practically. I can assist with discussions though.